bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #93: How to connect UPS 1.0 to SKR E3 DIP?

Meshchankin, 2020-07-30:
Sorry how to connect UPS 1.0 to SKR E3 DIP?
What should I write in #define POWER_LOSS_PIN ???
How to connect it to this main board ?

oldman4U, 2020-08-10:
Hi. This mainboard is a 1:1 replacement for the original mainboard which does not support the UPS. You will need another mainboard or a display which supports it.

Meshchankin, 2020-08-10:
> Hi. This mainboard is a 1:1 replacement for the original mainboard which does not support the UPS. You will need another mainboard or a display which supports it.

May be possible to use Filament sensor connector ?

oldman4U, 2020-08-26:
Hi. 

I use a BLTouch with this board but i am not aware if you can connect anything else to it. I would buy a TFT which supports all the features you would like to use.

oldman4U, 2020-09-27:
Please help the community and close the ticket once you do not need it anymore. Thank you

Meshchankin, 2020-10-01:
I have buy this ups because your official dealer at Aliexplress told me
that it will work with SKR E3 DIP.
Please add this information to UPS manual because it's important.
I will close

oldman4U, 2020-10-01:
I am not BTT but a user like you. Maybe you would like to ask your question in the Marlin repository because it finally has to work with it.

Bad documentation of all those features is a shame. 

Keeping a ticket open even after months is useless and makes the ticket system unusable.

Meshchankin, 2020-10-02:
Thank you!

Now I know this but your store doesn't know it.
They promise compatibility with ALL boards.
So please add this information to manual or stay it here.

oldman4U, 2020-10-02:
Again. It is not my store, and i have nothing to do with BTT or the dealer.

It is possible that there is a way to attach it and I do not know it, so it would be the best to contact the dealer who promised the compatibility and ask how to connect it. Once you know it, and you want to share what you learned, you could let us know.

This would be great

Thank you

oldman4U, 2020-10-09:
So have you contacted the dealer and have you got an answer?

Please let me know

closed 2020-10-13 by Meshchankin