bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #96: Pre-compiled firmware-A4988 has all 3 axes reversed

AlexanderAmelkin, 2020-08-28:
I have just installed E3 DIP v1.1 into my stock Ender 3, added four A4988 stepsticks, and flashed the pre-compiled firmware found here:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/blob/master/Firmware/firmware-A4988.bin

When I do Auto Home, my caret starts moving to the right, away from the endstop. If I manually press the endstop switch, then the table starts moving towards me (away from the endstop). If I manually press the Y endstop, then the Z axis starts moving... to the top!

Looks like all three axes are reversed for A4988.

oldman4U, 2020-08-29:
Switch the cables 180 degrees or change in firmware.

AlexanderAmelkin, 2020-08-31:
I surely rebuilt my own version of Marlin from official Marlin bugfix-2.0.x branch, but this issue is just to let you know that your pre-build binary for A4988 is wrongly configured. All connections are as in stock Ender 3. There is no way to "switch the cables 180 degrees", the JST XH connectors on the motors and on E3 DIP are keyed, and it all worked properly with the stock Creality board.

AlexanderAmelkin, 2020-08-31:
To clarify further: I do not have any problems anymore. However, anyone using your binary will have problems, and if they aren't familiar with rebuilding Marlin, they will be frustrated.

oldman4U, 2020-08-31:
Hi

It is not my build - I am user like you....

In the meantime I bought 5 different types of Nema motors and every time
the cables did not work but I had to adjust them. At least the two inner
cables had to be twisted.

But please do not forget to close this ticket one day or even better would
be if you could correct the issue and start a PR.

Do you know how this is done?

um 20:53:

AlexanderAmelkin, 2020-08-31:
I can correct, but I see no point in doing this here. It must all be submitted into mainstream Marlin repo, and I will probably do so. I don't see any reason for a Marlin clone to exist here at all except for creating PRs against the mainstream repo.

At the same time I don't think that pre-compiled binaries must be fixed by me or you or any other user. It's an official BTT support repo and IMO they must take care of this for their users who aren't familiar with Platform.io, Marlin, and all the other programming stuff.

Since you're not a BTT employee, let's please not discuss how or when this ticket must be closed. This is definitely out of your or my area of responsibility and/or competence. As I said earlier, I do not have any problems whatsoever with my SKR E3 DIP board. The ticket is just to notify BTT of the problem they have with their pre-built firmware and a stock Creality Ender 3 hardware (not some random motors you bought from random places). Let's not pollute this ticket further please.

ali1493, 2021-12-01:
> I surely rebuilt my own version of Marlin from official Marlin bugfix-2.0.x branch, but this issue is just to let you know that your pre-build binary for A4988 is wrongly configured. All connections [...]

Hi, can you share the bin file that you create  I wasn't able to create my own build and I have the same problem with the A4988 step drivers?