bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #97: Suggestion: use bigger flash in SKR E3 DIP v1.2 or some next revision

AlexanderAmelkin, 2020-09-01:
Latest versions of mainstream Marlin have difficulties fitting in the 256KB of flash on STM32F103RCT6.
If possible, could you please consider using STM32F103R**D**T6 instead?

Deneteus, 2020-09-13:
@AlexanderAmelkin V1.1

[image]

AlexanderAmelkin, 2020-09-15:
@Deneteus, that picture just means that you've used the STM32F103RC_btt_512K target in Platform.io.
That doesn't add memory to your chip. Or maybe BTT is using a counterfeit STM32F103 chip marked as RCT6, but having 512K of flash on the contrary to the official ST documentation. The documentation says that RCT6 is 256K, while 512K is only available on RET6, see https://www.st.com/resource/en/datasheet/stm32f103rc.pdf page 135:

[image]

AlexanderAmelkin, 2020-09-15:
Thanks for the hint though. I'll check if my board is using that counterfeit chip too and can accommodate more than 256K of progmem.

oldman4U, 2020-09-17:
My experience is, that it works but that the bootloader only accepts uploads of app. 220k.

I could imagine that BTT has chosen this CPU with the limitations in mind and offers other boards with better specs for a higher price.

AlexanderAmelkin, 2020-09-23:
They don't say it anywhere, but after thoroughly inspecting my SKR E3 DIP v1.1 I found that the chip is actually already STM32F103R**E**T6. All the media said it was RCT. I'm closing this issue.

closed 2020-09-23 by AlexanderAmelkin

oldman4U, 2020-09-23:
Thats why all the different firmware with 512k is available. But keep in mind that some users reported issues like stopped printing and interrupted movement when using 512k. I went back to 256 because the bootloader limits the max size of the firmware anyway. 

Thank you for closing the ticket 

Happy printing

AlexanderAmelkin, 2020-09-25:
It's a shame that the source of bootloader is not available. Or did I just fail to find it?