bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #99: Feature Request: Update firmware compilation instructions & Add BLTOUCH Setup to manual

Deneteus, 2020-09-14:
I have a couple of suggestions.
1. Every time there is a new revision there should be a PDF manual that matches that revision.
2. Every revision should have the instructions for how to build the current version of Marlin as the board names have changed.
3. Break out each Revision into its own folder section /bigtreetech/BIGTREETECH-SKR-E3-DIP/V1.0/
4. Add pictures on how to connect the BLTOUCH to the board and the pinout order to the manual.

There are too many guides with the wrong information so this will help your customers.

oldman4U, 2020-09-14:
Hi. 

This would be great. You know that all this is open source and therefore done by the community. You are part of the community, so it would be great if you could do this.

Thank you

👍🏻

oldman4U, 2020-09-17:
Hi

What is the status of your project?