bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #101: Hotend temperature setting with this board and TFT35 V3.0

Grayfox2005, 2020-12-03:
I can't seem to get above a 240c setting. It wants to advance higher but then it goes back to 240c in touchscreen mode but in marlin mode it wont advance any higher than 240c. I've checked the firmware of both of them and can only find where it is defined at 275c maximum on the TFT. I don't know if its a setting in this board or the TFT.

Grayfox2005, 2020-12-04:
Well I found someone having a similar issue with the setting. Not the same board but same issue. Seems the runaway protection has something to do with it so I looked into the firmware code and my hot end   #define HEATER_0_MAXTEMP 255 and all the other temps were 275 so I changed it to 275, compiled it, and flashed it. Now I don't have a connection to the touch mode on the TFT35. It says "No Printer Attached" Marlin mode is fine and it did change the max temp I could adjust to 260. 

Why is it I'm always taking my chance when I make a new copy of the firmware to flash? My VS code is always fxcked up. It never lets me start in right away with compiling code like its updating files behind the scenes. This is the same firmware code I've been using and it worked before but now it doesn't.

If you look at this link it does follow what they found to be the problem with the temp setting. https://www.reddit.com/r/3Dprinting/comments/ih4hzf/printer_refuses_to_set_temperature_as_high_as_i/

oldman4U, 2020-12-04:
Baud rate in Marlin and TFT firmware does not match.

Grayfox2005, 2020-12-04:
Unfortunately that wasn't the problem. I've ran into that before and made sure I double checked it. I found a copy of my previous firmware and put that back in and it works like a charm. I only made that one small change in the firmware code and all else was the same. 

Something is happening with VSCode in the background like it's updating the files in the project and I always have to wait until its done. So I'm taking it for granted that VSCode is going to the internet somewhere and is updating the files used in the project every time I start it up. Whenever I use the Platform IO tool bar at the bottom it comes back and tells me _command 'platformio-ide.build' not found_ until its done doing what it's doing. The build folder keeps filling with a bunch of folders of board technologies and then when it's finished they are all gone again. I suppose I'll have to go to reddit to discuss what's going on and see if I can fix that from there. It's probably something to do with Platform IO. I've uninstalled VSCode and reinstalled it several times with no luck. I have it installed on my laptop as well and I don't get that behavior on it. I'm going to copy the project onto it and see what happens. Another day spent troubleshooting the tools I need to work the printer... good God! 
But if I can get a working firmware the above comment is the fix So I'm pretty sure I'll be closing this issue today.

closed 2020-12-05 by Grayfox2005

oldman4U, 2020-12-05:
Had many VSC issues too. Even it has been working for weeks you never know if it will work tomorrow. 

Happy printing

Grayfox2005, 2020-12-05:
Thanks and just for a heads up I got the latest Marlin 2.0.7.2 firmware to compile and install and work fine with the display. It already had the change done to it so I can set my nozzle temp up to 260. Don't know why 2.0.5 firmware was set lower.