bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #102: Noisy X motor with SKR marlin 2.0.7.2 release

Grayfox2005, 2020-12-05:
I just installed the latest firmware from this board and noticed my X axis motor has a scratchy noise like something is rubbing but its coming out of the motor. It was fine with the previous release, I couldn't hear any of the motors. Seems funny it just affects the X. I disabled hybrid threshold because that wasn't enabled in the previous release and that was no help. It sounds like a critter in the ceiling scratching around lol. Just wondered if there is something else I could try to quiet it?

Grayfox2005, 2020-12-06:
Actually it could be all the motors but only the x seems prominent.

oldman4U, 2020-12-14:
Which drivers with which setting?

Most likely a Marlin TMC issue.

Grayfox2005, 2020-12-14:
TMC2209 drivers same settings as the earlier version of Marlin. I put the old one back in and it was definitely gone so its something somewhere in the latest. Just waiting for others to chime in their results.

oldman4U, 2020-12-14:
Marlin often includes new version of the TMC library. I would recommend to
compile your own firmware, which allows to change the libraries  used.

Grayfox2005, 2020-12-14:
Thanks I'll look and see the differences in old vs the new. I always compile my own version because I have much more to my printer than is included in the pre-compiled version. I see they are including a lot more in the latest version in controlling the settings of the board and extra hardware which is nice.

oldman4U, 2020-12-14:
Great.

Please let me know in case you find something.

Happy printing

Grayfox2005, 2021-01-07:
Guess I'll close this. It's been a long time with no other complaints so it must be an issue on my end.

closed 2021-01-07 by Grayfox2005