bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #103: BigTreeTech SKR E3 DIP v1.1

DaddyCRO, 2020-12-11:
Recently purchased BigTreeTech SKR E3 DIP v1.1 with four TMC5160 drivers. Everything works like a charm (very quiet drivers).
The board is built-in in Creality CR-20 Pro 3D printer.
Also purchased Bigtreetech TFT24 v1.1 touch screen display. It is working correctly with E3 DIP board in Marlin mode.
However, unable to set it up to work in touch screen mode. Have a message "no printer attached" in this mode.
Marlin is last version (2.0.7.2?). Also, TFT24 is flashed with latest firmware from GitHub.
Both cables are correctly connected (checked pinout several times).

Any suggestions? Is this board able to work with TFT24 in touch screen mode?
Thnx for help/sugestions.

oldman4U, 2020-12-11:
Hi

It works

Send a picture which shows the cable and what is the baud rate defined in Marlin and TFT?

DaddyCRO, 2020-12-11:
Hi,

please find settings bellow...

SKR E3 DIP Configuration.h:
---------------------------------
#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1
#define BAUDRATE 115200
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define CR10_STOCKDISPLAY

BTT TFT24_V1.1 config.ini:
---------------------------------
# Options: [Unified Menu: 1, Classic Menu: 0]
unified_menu:0
#### Baudrate
# Options: [2400: 0, 9600: 1, 19200: 2, 38400: 3, 57600: 4, 115200: 5, 250000: 6, 500000: 7, 1000000: 8]
baudrate:5
#--------------------------------------------------------------------
# Marlin Mode Settings (Only for TFT24_V1.1 & TFT28/TFT35/TFT43/TFT50/TFT70_V3.0)
#--------------------------------------------------------------------

#### Default Mode
# Options: [0: Marlin Mode, 1: Touch Mode]
default_mode:1

#### Keep Serial Always ON
# Keep UART (Serial communication) alive in Marlin Mode
# Allow seamless OctoPrint UART connection to the TFT's UART/serial expansion port no matter which mode the TFT is in.
# Options: [0: Disabled, 1: Enabled]
serial_always_on:1

#### Default Marlin Mode Background & Font Color Options
# Options: [ WHITE: 0,  BLACK: 1,  RED: 2,  GREEN: 3,      BLUE: 4,       CYAN: 5,  MAGENTA: 6,    YELLOW: 7,
#           ORANGE: 8, PURPLE: 9, LIME: 10, BROWN: 11, DARKBLUE: 12, DARKGREEN: 13,    GRAY: 14, DARKGRAY: 15]
# Or set the color(RGB888 format) hex value directly(start with “0x”).
# Such as: 0xFF0000 : Red, 0x00FF00 : Green, 0x0000FF : Blue
marlin_bg_color:1
marlin_fn_color:8

#### Marlin Mode show title
# Options: [enable: 1, disable: 0]
marlin_show_title:1

# Run Marlin Mode in Fullscreen
# Options: [enable: 1, disable: 0] [Disabled. RECOMMENDED FOR TFT24]
marlin_fullscreen:0

# Select marlin type
# Options: [LCD12864: 1, LCD2004: 0]
marlin_type:1

#### Text displayed at the top of the TFT in Marlin Mode.
marlin_title:LCD12864 Simulator

Also, as suggested, all these options are enabled in Marlin firmware:

#  Note: Ensure that following options are enabled in Marlin firmware
#
#  M115_GEOMETRY_REPORT (in Configuration_adv.h)
#  M114_DETAIL  (in Configuration_adv.h)
#  REPORT_FAN_CHANGE (in Configuration_adv.h)
#  EMERGENCY_PARSER (in Configuration_adv.h)
#  SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
#  HOST_ACTION_COMMANDS  (in Configuration_adv.h)
#
# For printing from onboard SD
#
#  SDSUPPORT (in Configuration.h)
#  AUTO_REPORT_TEMPERATURESS (in Configuration_adv.h)
#  AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
#  LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
#  SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)

Thank you for support!
P.S. I do not know why there is so big font in reply., so sorry for that...

DaddyCRO, 2020-12-11:
[image]
Also, please finf picture of TFT24 and cables...

oldman4U, 2020-12-11:
I am wondering that Marlin works.

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Has to be changed to
//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

Try also to change the serial port from 2 to 0.

DaddyCRO, 2020-12-11:
Tried with both settings already.
If I comment //#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, it is also not working. Then for LCD/TFT section stays only uncommented "#define CR10_STOCKDISPLAY". Does a board needs (somehow) an information that TFT screens is connected to the board?

If I set serial port to "0", Marlin displays error during building firmware:
#error "SERIAL_PORT must be -1 or from 1 to 5. Please update your configuration."

oldman4U, 2020-12-11:
Do you have a picture of the mainboard and how the cable looks like there
also.

DaddyCRO, 2020-12-11:
Not in this moment. I will tomorrow (in Croatia is almost midnight :-) ), open a printer, and take a picture.
I saw here that there is a difference between v1.0 and v1.1 in pins:
[image]
So I follow this scheme to connect a cable...
BOARD	TFT
-----------------
 RST         RST
 RX2         TX
 TX2         RX
 GND	GND
 +5V	+5V

I even changed pins on the board side cable connector to match RX-RX and TX-TX but without success...

oldman4U, 2020-12-11:
It is after midnight here in Austria 😁

DaddyCRO, 2020-12-11:
:-) Greetings then from Croatia to Austria... I am working for one Austrian company here in Zagreb...

oldman4U, 2020-12-11:
Greetings back. 👍🏻

DaddyCRO, 2020-12-11:
Forgot to mention that I also compared "pins_BTT_SKR_E3_DIP.h" and "pins_BTT_SKR_PRO_common.h" files in VSC.
In "pins_BTT_SKR_E3_DIP.h" file is missing section (exists in "pins_BTT_SKR_PRO_common.h" which uses other BTT board, except mini):

#if IS_TFTGLCD_PANEL
  #if ENABLED(TFTGLCD_PANEL_SPI)
    #define TFTGLCD_CS                      PG10
  #endif

Do not know if that affects to board TFT compatibility...

oldman4U, 2020-12-12:
The combination of mainboard and TFT works out of the box without changing any pins file. 

Connection
Serial port
Baud rate

DaddyCRO, 2020-12-12:
[image]

DaddyCRO, 2020-12-12:
[image]

DaddyCRO, 2020-12-12:
[image]

DaddyCRO, 2020-12-12:
[image]

oldman4U, 2020-12-13:
OK, so the cabling is right.

Assuming the cable is not broken and also that the SKR and the TFT are OK it has to be in software.

So I checked again Marlin and I have the following defined:
#define SERIAL_PORT 2
//#define SERIAL_PORT_2 -1 //do not need USB so it is uncommented
#define BAUDRATE 115200

With this setting and baud rate of the TFT also set to 115200, this has to work. Check the baud rate in the corresponding touchscreen menu.

DaddyCRO, 2020-12-13:
It is strange that I cannot enter menu in touch screen mode. Long press on screen leads to a menu for choose between Marlin and touch screen mode, so I cannot check baud rate on TFT24. Yesterday I flashed new firmware versio on TFT without problem... In config.ini for TFT baud rate is set to 115200... Single press on any icon on screen has no response...
Will try to comment line and rebuild Marlin...
Thank you for your help...

oldman4U, 2020-12-13:
Wait. You can not press a button on the touchscreen??

Make a empty textile named reset.txt and put it on the sd card. put it in the TFT and reset the TFT. Let me know

DaddyCRO, 2020-12-13:
I did. After restart, it brings a menu for touch screen calibration...
I successfully calibrated the screen, and now I can press on icons. So thank you for this tip!
Checked baud rate on TFT24 and it is correctly sated up to 115200...
Commented //#define SERIAL_PORT_2 -1 line, and building firmware...

DaddyCRO, 2020-12-13:
Just did. But I still have "No printer attached!" message on screen.
Do not know if it has some matter, but I am compiling in STM32F103RE_btt_USB mode...
I have STM32F103RET6 chip on board...

oldman4U, 2020-12-13:
STM32F103RC_btt is my defaults_envs

DaddyCRO, 2020-12-13:
[image]

DaddyCRO, 2020-12-13:
Just build "STM32F103RE_btt" and flash. Still the same... No connection...

oldman4U, 2020-12-13:
Replace the cable. Can you make a new one?

DaddyCRO, 2020-12-13:
I will check it with my instrument (pin by pin connection)... I have digital multimeter at home...
But it is brand new. I got it with my TFT24...

oldman4U, 2020-12-13:
Ok.

oldman4U, 2020-12-14:
What happens when you disconnect the 10pin cables and only keep the 5pin cable connected?

DaddyCRO, 2020-12-14:
Hi. It powers up normally, butin touch screen mode still reports error message "No printer attached!" Got back from office, will check the cable immediately...

oldman4U, 2020-12-14:
Serial port 2
Baud rate the same

If those two are correct it has to be hardware.

DaddyCRO, 2020-12-14:
Just checked the cable: all pins have connection cable seems OK to me.

DaddyCRO, 2020-12-14:
One thing that bothers me is cable pinout, I am not sure if it is correct...

At the board, top most PIN is RST, and it is connected with RST pin on TFT24
next pin below on the board is RX2, it is connected to the TX2 pin on TFT24
next pin below on the board is TX2, it is connected to the RX2 pin on TFT24
next pin below on the board is GND, it is connected to the GND pin on TFT24
next pin below (bottom one) on the board is +5V, it is connected to the +5V pin on TFT24

**Does TX2 and RX2 pins needs to be connected with corresponding TX and RX pins on TFT24?**
Or it should be as it is connected right now (describe earlier)?

DaddyCRO, 2020-12-14:
Just swapped the cable pins so TX on the board points to the TX on TFT24, and RX on the board points on RX on TFT24, but without success. Still no conection...

DaddyCRO, 2020-12-14:
What about serial port configuration number? I tried with 1 and 2: should I try with other numbers from the range (0 - 5)?

oldman4U, 2020-12-14:
I used the cable without modifications. Connections are straight.

I asked you to disconnect all other cables because the TFT has to work in
touchmode with 5pin cable only.

oldman4U, 2020-12-14:
Your motherboard requires 2.

DaddyCRO, 2020-12-14:
So it is HW issue: either board, TFT or both...

oldman4U, 2020-12-14:
Guess so. Or cable. Or something in pins which you changed falsly. If you
send me Dropbox or something else link I can send you the firmware of
mainboard and TFT.

DaddyCRO, 2020-12-14:
Can you send it in e-mail in zip file?

oldman4U, 2020-12-14:
Yes

DaddyCRO, 2020-12-14:
Then please send it to: les.darko@gmail.com
Thank you...

oldman4U, 2020-12-14:
[image]
[image]
[image]

DaddyCRO, 2020-12-14:
Nice!

oldman4U, 2020-12-16:
Please help the community and close the ticket once you do not need it anymore. Thank you 

Happy printing

DaddyCRO, 2020-12-16:
Just to inform that I finally found out where the problem was... Thank you "oldman4U" for your support.
I added a filament run out sensor on the board, and somehow line 1183 in "Configuration.h" file stays uncommented:

#define FIL_RUNOUT_PIN 2 // Creality CR20-S stock sensor

After commenting this line, TFT works perfect! It seems that there is some sort of collision between serial port communication and this parameter.

Will close the ticket, since problem (for me) is solved...

closed 2020-12-16 by DaddyCRO