bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #116: Ender-3 screen not truning on

daniframinan, 2022-02-12:
HI, I have a Creality Ender-3 with a SKR-E3-DIP-V1.0, and it's screen has stopped truning on. Not only that, but it's impossible to communicate with it via USB, ad the Status light is always on. I believe it's a hardware issue, but I would like confirmation. If it is a hardware problem, does BigTreeTech offer any sort of fixing, replacing or refunding program? Thanks.

daniframinan, 2022-02-12:
I bought the board on the 31st of May 2021

radek8, 2022-02-12:
Don't expect support for BTT here.
If he answers, he'll tell you to deal with the dealer.
When did the problem arise? Didn't you add new HW?
Didn't that happen after adding the leveling probe?

daniframinan, 2022-02-12:
Hi, the problem arose last Tuesday, after I tried using the Change filament function on Marlin 2

I had already had the probe working for about half a year by then

The only new hardware was a Hemera extruder I had before I bought the board, and a TMC 2209 replacing the extruder's 2208 after it broke, but that was before Christmas.

The problem is the dealer (on Amazon) has stopped selling them since then, so I can't contact them. I've written a letter to BTT, but I also doubt they will respond.

Thanks.