bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0 #118: Cannot compile in Visual Studio Code

YapCK45, 2023-05-30:
I downloaded BIGTREETECH-SKR-E3-DIP-V1.0-master.zip and unzipped it in VSC. Checked all the changes needed and noted to change default_envs in platformio.ini to STM32F103RC_btt. When compiling got the following error message:

File "D:\BTT Firmware\BIGTREETECH-SKR-E3-DIP-V1.0-master\Firmware\Marlin-2.0.7.2-SKR-E3-DIP\buildroot\share\PlatformIO\scripts\STM32F103RC_SKR_MINI.py", line 6:
    for define in env['CPPDEFINES']:

Screen shot here:
[image]

Please help.

H-DieterK, 2023-05-30:
Hi YapCK45,VSC often won’t compile with long file names. Try copying the folder named Marlin-SKR-2.0.7.2-SKR-E3-DIP and pasting it into a new simply named blank folder onto your Desktop to work from.Not sure why you are referencing an SKR-E3-Mini which will likely have a different CPU environment name. My own BTT SKR E3 DIP V1.1 is an ARM STM32F103RTE6, which I also has compilation errors on the same version of Marlin which I could never customize configs.h and .avh to without failing! Upgraded board to SKRE3Turbo which is also environmentally fussy but much better. Likely Female!I suspect that the MINI’s env won’t work!Sorry, can’t help further.Best of luck 🤞 CheersHansI suspect Sent from Hans' iPhone

YapCK45, 2023-05-31:
Thank you, @H-DieterK for your comment: 'Not sure why you are referencing an SKR-E3-Mini which will likely have a different CPU environment name'. I am puzzled too as this was in platformio.ini  and as below:

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = STM32F103RC_btt
include_dir  = Marlin
.
.
.
[env:STM32F103RC_btt]
platform          = ${common_stm32f1.platform}
extends           = env:STM32F103RC
extra_scripts     = ${common.extra_scripts}
  buildroot/share/PlatformIO/scripts/**STM32F103RC_SKR_MINI.py**
build_flags       = ${common_stm32f1.build_flags}
  -DDEBUG_LEVEL=0 -DSS_TIMER=4
monitor_speed     = 115200

I am using the shortest path and using the folder named Marlin-SKR-2.0.7.2-SKR-E3-DIP. It made no difference.